adafruit/RGB-matrix-Panel #18: Incompatibility with SoftwareSerial library

mdauria, 2015-03-29:
Enable Timer1 interrupt causes an instability of the SoftwareSerial library :
TIMSK1 |= _BV(TOIE1); // Enable Timer1 interrupt (in "void RGBmatrixPanel::begin(void)" function)

So I can't use a bluetooth module with the RGB matrix : Timer 1 causes the appearance of random characters at the bluetooth reception...

shafiqurrehman, 2016-12-22:
Buddy, Did you find any solution to the timer issue? 

Same timer is messing up I2C communication as well.

Thanks,

arduibag, 2016-12-25:
Hi,

I juste changed the Bluetooth module (another reference), then it works !?!

Mike

obbaeiei, 2019-04-04:
@arduibag hi, which module you have changed

marcmerlin, 2019-04-04:
can you connect the BT module to a serial port with UART?
Using 2 pieces of software that do bit banging (i.e. this lib + softwareserial) is not super likely to work unless timing happens to work half by luck

obbaeiei, 2019-04-04:
@marcmerlin I'm connect with UART. It still random characters

marcmerlin, 2019-04-04:
Ah, in this case this is not what I expected. All I can think of is that this driver is taxing your chip so much (I see no info on what chip is being used) that it can't empty the UART buffer quickly enough.
If that's the case, I recommend you upgrade to better hardware with SmartMatrix (teensy with SmartMatrix shield, or ESP32 if you're ok doing your own wiring)

obbaeiei, 2019-04-04:
Thank you for your recommendation. I'm using mega2560 with AT-09 BLE