adafruit/RGB-matrix-Panel #26: Spurious .github folder warning

per1234, 2016-06-08:
The .github folder added in https://github.com/adafruit/RGB-matrix-Panel/commit/14d8b3c78831c4a03fd67a7a5cb6d860bc37e3da(and apparently to all other Adafruit libraries) causes the warning:
```
WARNING: Spurious .github folder in 'RGB matrix Panel' library
```
with Arduino IDE 1.6.6 or greater on every compile, whether the library is included or not. This is annoying and causes confusion for beginners. I would recommend putting the template files in the repository root instead as https://help.github.com/articles/creating-an-issue-template-for-your-repository/ states is allowed.

ladyada, 2018-05-25:
@per1234 hiya, why not just just Arduino ignore .github the same way it ignores .git?

per1234, 2018-05-25:
The .github folder has been ignored since arduino-builder 1.3.19 / Arduino IDE 1.6.10:
https://github.com/arduino/arduino-builder/commit/18470b99daca196257205eae9396a1f74f2bd6fe

Actually it was done at your request: https://github.com/arduino/arduino-builder/issues/70#issuecomment-227041842
Thanks!

So at this point the issue only exists for people using outdated versions of the Arduino IDE (1.6.6-1.6.9). Since it's only a warning anyway I'd say the issue can definitely be considered resolved. I just forgot about this issue and so I never closed it.

closed 2018-05-25 by per1234

ladyada, 2018-05-25:
oh lol, i wasnt watching this repo and just did an issue sweep. well, GOOD WORK PAST US! :D