adafruit/RGB-matrix-Panel #29: I2C communiation

shafiqurrehman, 2016-12-22:
Hello,

looks like this library manipulates the clock or timer of Arduino which in turn messes up all communication capabilities of the Arduino boards.

I have tried I2C and serial communication with other Arduino without this library and it works flawlessly. but as I introduce RGB library in the code, all type of communication literally stops.

Is there any other library that needs to be used with this Adafruit RGB library for I2C and serial communication?

Thanks,

PaintYourDragon, 2016-12-22:
Library is "squeezing blood" from an 8-bit AVR as it is, doing horrific things with timers and interrupts. If anything more sophisticated that basic images or animation is needed, I'd suggest switching to a more capable dev board. Teensy 3.X, Raspberry Pi and BeagleBone all have much improved libraries available and no problem handling additional tasks.

ShafiqUrRehman, 2016-12-22:
Thanks Dragon.....

I have tried it on Raspbery and it worked there.... I am trying to keep the project as simple as possible so I will try Teensy.

Thanks,

closed 2019-10-26 by ShafiqUrRehman