adafruit/RGB-matrix-Panel #31: Support for MKR1000

LRagji, 2017-04-17:
Hi, I would really love to see support for MKR1000 as that board is powerful in all aspects any support for same anything 

This is not a bug its a feature request?

ladyada, 2018-08-22:
we support samd21 now

closed 2018-08-22 by ladyada

bgaebel, 2018-11-04:
Are you sure the support of samd21works well?
I have two MKR1000 and when I just add matrix.begin() to my sketch, the boards hangss up and will also disappear from Windows Device Manager (error code 43)

any ideas?

thanks

ladyada, 2018-11-04:
we havent tested the MKR1000 - that may have a different pinout setup. it works with our samd21 boards for sure!

bgaebel, 2018-11-05:
Dear ladyada,

it was my fault! I missed the additional parameter *pinlist of the contructor.
Library works as expected for the MKR1000

*thumbs_up

best regards

LRagji, 2018-11-06:
@bgaebel can you share the pinouts/connection diagram for same?, also the code part of it if possible.

bgaebel, 2018-11-06:
Sorry, I were to fast with my pleasure.
Just using matrix.begin() works, but until I want to draw any Pixel, the entire board hangs up.

How to figure out, what's going wrong?

On my MKR1000 works an ATSAMW25H18... and there I see the mistake: It's not a SAMD21, it's a SAMW51... 
GOD DAMN IT!

LRagji, 2018-11-08:
I think this means the defect is still open, as it talks about MKR1000 support?

ladyada, 2018-11-08:
where we talk about MKR1000 support? we'll remove it