adafruit/RGB-matrix-Panel #35: How to change the assignment pins in this library?

wavewolf, 2018-02-27:
Hi, I'm using two different libraries in my arduino code and my problem is that one of them lets me assign the pins in the same code that I do, but the other library(RGB-matrix-Panel) assumes the digital pins privately and I dont know how to change the assignment of the pins. I need to change the assignment of the digital pin 2 to the digital pin 12.

per1234, 2018-02-27:
Also posted at http://forum.arduino.cc/index.php?topic=531890.

PaintYourDragon, 2018-02-27:
Pins 2-7 cannot be reassigned. The code uses PORT-wide writes for the 6 color data lines, it's the only way to operate fast enough on an AVR. Other various control signals can be assigned to other pins (as passed to the constructor), but those 6 are simply not negotiable.

closed 2018-02-27 by PaintYourDragon

wavewolf, 2018-02-27:
ooooh.... Perfect solution. "It can be possible", thanks....

wavewolf, 2018-02-27:
I got it. I have neutralized the red color at the top of the screen (I didn't need it) and I already have the digital pin 2 FREE.