adafruit/RGB-matrix-Panel #42: cannot run PanelGFXDemo_16x32 example

bagustyo92, 2018-07-09:
no function in RGBmatrixPanel.cpp named drawRGBitmap() that you used on PanelGFXDemo_16x32.cpp file.

- Arduino board:  mega2560
- Arduino IDE version (found in Arduino -> About Arduino menu):  Platform IO based on Arduino IDE 1.8.5
- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too): unzip the zip file and import to lib platform IO folder

bagustyo92, 2018-07-09:
typo in my comment, but surely in the real code is exactly the same that you write, sorry for sure, still not work

[image]

bagustyo92, 2018-07-09:
solved!
the library that i used is not up to date because some reason, thanks

marcmerlin, 2018-07-09:
Ok, this was the next things I was going to suggest :)   Glad it's solved. 
Can you close the issue please?

closed 2018-07-10 by bagustyo92