adafruit/RGB-matrix-Panel #44: Running on Matrix 96x16

CrackerNo1, 2018-11-10:
[image]

I tried to connect 3 module P10 Full Color as shown as the image but it does not work.
"RGBmatrixPanel matrix(16, A, B, C, CLK, LAT, OE, true, 96);". This is my config module P10.
Please help me to make it work correctly.
Thank you so much!
```
#include <RGBmatrixPanel.h>

#define CLK  8
#define OE   9
#define LAT 10
#define A   A0
#define B   A1
#define C   A2

RGBmatrixPanel matrix(16, A, B, C, CLK, LAT, OE, true, 96);
int16_t textX = matrix.width();

void setup() {
  matrix.begin();
  matrix.setTextWrap(false); // Allow text to run off right edge
  matrix.setTextSize(2);
}

void loop()
{
  char text[] = "Github.com";
  int16_t textMin       = sizeof(text) * -12;
  matrix.fillScreen(0);
  matrix.setTextColor(matrix.Color333(255, 255, 255));
  matrix.setCursor(textX, 1);
  matrix.print(text);
  delay(30);
  if ((--textX) < textMin)
  {
    textX = matrix.width();
  }
  matrix.swapBuffers(false);
}
```

bgaebel, 2018-11-10:
What board do you use?
As I know, this library still doesn't support chained panels =(
I'm in the same situation!
I have some of the 64x32 P4 Panels and want to get them working on a Feather M0.
This board should have enough ressources for this job.

CrackerNo1, 2018-11-10:
> Which board do you use?
> As I know, this library still doesn't support chained panels =(
> I'm in the same situation!
> I have some of the 64x32 P4 Panels and want to get them working on a Feather M0.
> This board should have enough ressources for this job.

Arduino is the board which I used, I just scroll short text from Bluetooth module, I think it enough memory to my project.
Do you know any library support chained panels??

ladyada, 2018-11-11:
hiya you can keep chattin' in this thread, closing cause its not really an issue :)

closed 2018-11-11 by ladyada