adafruit/RGB-matrix-Panel #46: text split and shifted

sebasobotka, 2018-11-28:
Hi,I am using Hub75 panel and  I've got an issue with drawing text. It seems that something is mixed up. 

`matrix.setCursor(1, 0);   // start at top left, with one pixel of spacing
matrix.setTextColor(matrix.Color333(0,0,255));
matrix.print('2');
`
And the resultis on the picture. Could you point me to the right direction?
[image]

ladyada, 2018-11-28:
hi did you buy the panel from adafruit? if not, please contact the seller for support - thank you :)

sebasobotka, 2018-11-29:
hi i didn't buy the panel from adafruit but i fixed it. we can close the issue.

closed 2018-11-29 by sebasobotka

DoomHammer, 2022-11-16:
> hi i didn't buy the panel from adafruit but i fixed it. we can close the issue.

Hi there!

Can you share how did you fix it? :)

sebasobotka, 2022-11-16:
@DoomHammer I don't remember ;) It was 4 years ago. Maybe I've got somewhere the code. I will try to find it later.

DoomHammer, 2022-11-16:
If you could, it'd be more awesome than paprykarz szczeciński!

sebasobotka, 2022-11-24:
@DoomHammer sorry but I didn't find anything. What kind of issue do you have with that?

DoomHammer, 2022-11-24:
Basically, I have those 32x16 panels with 3 address lines. When I interface them and try to light a single LED, I see 4 of them turned on. I can only guess this may be related to scanning or something, but I have no idea how to configure it with `RGBmatrixPanel`.

sebasobotka, 2022-11-24:
@DoomHammer I found something ;)

First of all you need to check the pin description on the panel. I've got two of them with only A and B pins (not C and D) and two with C pin (and these are working).
Second thing is type of scaning mode. There are different type of panels. I've got 1/4 scan type. 
Third thing is type of Arduino. I was using Arduino Uno.

There are 3 steps:
1. pin connections like here
https://learn.adafruit.com/32x16-32x32-rgb-led-matrix/connecting-using-a-proto-shield
R0 - 2 (or R1) 
G0 - 3
B0 - 4
R1 - 5 (or R2)
G1 - 6
B1 - 7
A - A0
B - A1
C - A2
CLK - 8
LAT - 10
OE - 9
GND - GND

2. software from here
https://github.com/adafruit/RGB-matrix-Panel
After these 2 steps you should be able to run some examples from adafruit RGB matrix panel library. I had some issues with fonts and I wasn't able to print text on the panel. 

3. Here is a solution for my problem:
https://forum.arduino.cc/index.php?topic=503416.0
You need to download two files RGBmatrixPanel.cpp and RGBmatrixPanel.h and change &gamma to &gamma_table inside the RGBmatrixPanel.cpp

Here is an example of working:
https://www.instagram.com/p/BrAQxpkim95/