adafruit/RGB-matrix-Panel #53: Heat development of 64x32 RGB LED Matrix

lorach77, 2019-03-14:
Hi there,

I am using 64x32 RGB LED Matrices in a project where heat development (even some degrees) is a problem. When the LED matrix is connected to the power supply (no LED are lit), there is a heating up of the matrix around 4-7 degree, especially at the location of the micro-controllers. This is expected for me.
In my project, the matrix has to work only for short periods of time; e.g. 1h break, 2s lighting up, 5 min break, 2 s lightning up, ... So I though, I just interrupt the power supply to the matrix (no power, no warming) during the "breaks".

I now realized, that if the matrix is connected to my Arduino mega and NOT connected to the power supply, the connection of the IDC cable to the matrix warms up very fast (see [image](https://photos.app.goo.gl/axKjvMW1nv2dhCxb8); FLIR of the LED matrix. Left: power supply not connected, right: power supply connected)

1. Why is this connection only warming up when the power supply is not connected? (I don't have a lot of electronic background) 

It only warms up after the begin.matrix() function is called. I measured the pins of the Arduino. The 4 analog pins show 2.5 V, the CLK, LAT, OE around 0.4 V (?). I looked to the code of the RGB-matrix-Panel library but could not really make sense of it.

2. Is there a simply (software) solution? 

E.g. a simply way to write a matrix.stop() function (so i could use them before a "break"). Or can I just manually set the (analog) pins to 0V and up again before I need the matrix?

I hope I could explain the problem and thanks for your help. Best,
lorach

ladyada, 2019-03-14:
hiya @lorach77 where did you purchase the panel?

lorach77, 2019-03-14:
Boxtec AG Switzerland, see: [here their site](https://shop.boxtec.ch/64x32-rgb-led-matrix-3mm-abstand-p-42765.html)

Since they link to Adafruit, I guess it is the same matrix and they are a reseller.

ladyada, 2019-03-14:
hmm they're not listed as a distributor for this item
https://www.adafruit.com/product/2279#distributors
i think its not from adafruit - perhaps they just link to us - please contact them for support - they have a team of engineers who can help you! :)

lorach77, 2019-03-15:
Thanks for your response. 
Do you know that this problem does not occur with a original Adafruit matrix? If this is the case, I am thinking of buying a new one instead of troubleshooting.

ladyada, 2019-03-16:
not sure, but you need to have the panel ground always powered