adafruit/RGB-matrix-Panel #55: Support for FM6126A

ntsarb, 2019-06-16:
Thank you for opening an issue on an Adafruit Arduino library repository.  To
improve the speed of resolution please review the following guidelines and
common troubleshooting steps below before creating the issue:

- **Do not use GitHub issues for troubleshooting projects and issues.**  Instead use
  the forums at http://forums.adafruit.com to ask questions and troubleshoot why
  something isn't working as expected.  In many cases the problem is a common issue
  that you will more quickly receive help from the forum community.  GitHub issues
  are meant for known defects in the code.  If you don't know if there is a defect
  in the code then start with troubleshooting on the forum first.

- **If following a tutorial or guide be sure you didn't miss a step.** Carefully
  check all of the steps and commands to run have been followed.  Consult the
  forum if you're unsure or have questions about steps in a guide/tutorial.

- **For Arduino projects check these very common issues to ensure they don't apply**:

  - For uploading sketches or communicating with the board make sure you're using
    a **USB data cable** and **not** a **USB charge-only cable**.  It is sometimes
    very hard to tell the difference between a data and charge cable!  Try using the
    cable with other devices or swapping to another cable to confirm it is not
    the problem.

  - **Be sure you are supplying adequate power to the board.**  Check the specs of
    your board and plug in an external power supply.  In many cases just
    plugging a board into your computer is not enough to power it and other
    peripherals.

  - **Double check all soldering joints and connections.**  Flakey connections
    cause many mysterious problems.  See the [guide to excellent soldering](https://learn.adafruit.com/adafruit-guide-excellent-soldering/tools) for examples of good solder joints.

  - **Ensure you are using an official Arduino or Adafruit board.** We can't
    guarantee a clone board will have the same functionality and work as expected
    with this code and don't support them.

If you're sure this issue is a defect in the code and checked the steps above
please fill in the following fields to provide enough troubleshooting information.
You may delete the guideline and text above to just leave the following details:

- Arduino board:  **INSERT ARDUINO BOARD NAME/TYPE HERE**

- Arduino IDE version (found in Arduino -> About Arduino menu):  **INSERT ARDUINO
  VERSION HERE**

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too): **LIST REPRO STEPS BELOW**

ntsarb, 2019-06-16:
It would be great to see some support for LED matrix displays that rely on the FM6126A chip. Looks like the PxMatrix library  (for ESP8266) already supports it, but I was looking to use the display with an Arduino Mega 2560.

https://community.pixelmatix.com/t/smartmatrix-doesnt-support-fm6126a-driver-chips/421

I'm relatively new to Arduino development and I intend to look into the PxMatrix source code but I'd like to hear if there's already an interest/active work to port RGB Matrix Panel library to support FM6126A-based displays. Thanks you.

marcmerlin, 2019-06-16:
Howdy, so I guess you found my previous work looking around for support in making those panels work.
You may have found this script https://github.com/marcmerlin/SmartMatrix/blob/FM6126A/examples/FM6126A_reset/FM6126A_reset.ino
which can be ported to your processor if you so wish.
But keep in mind that adafruit provides libs and support for hardware sold by adafruit, and currently their panels do not have the FM6126A chip, so I believe your request will just get closed.
That being said, if I may, I recommend you switch from arduino mega to ESP32 and SmartMatrix + SmartMatrix::GFX, as an ESP32 is both cheaper and a lot more capable. In theory you need level shifters back from 3.3V to 5V, I use them, but apparently you can get by without them.

You can have a look at http://marc.merlins.org/perso/arduino/post_2019-04-01_SmartMatrix_-SmartMatrix-Shield-v4-for-Teensy_-ESP32-shield-with-level-shifter_-and-SmartMatrix_GFX.html#ESP32 if you'd like more info on how to use SmartMatrix with ESP32.
Good luck.

marcmerlin, 2020-01-24:
As another option, FM6126A is now natively supported in rPi.
I also wrote a driver that allows you to run arduino code on an rPi and send the output to RGBPanels: http://marc.merlins.org/perso/arduino/post_2020-01-01_Running-FastLED_-Adafruit_GFX_-and-LEDMatrix-code-on-High-Resolution-RGBPanels-with-a-Raspberry-Pi.html