adafruit/RGB-matrix-Panel #57: With 32x64 Panel Configuration there is no clock signal

rtek1000, 2019-09-10:
Hi,

I found a tutorial (link below) and when I add the number 64 the clock signal stops being done:

Not works:
RGBmatrixPanel matrix(A, B, C, D, CLK, LAT, OE, false, 64); // 32x64

Works (with duplicate image):
RGBmatrixPanel matrix(A, B, C, D, CLK, LAT, OE, false); // 32x32

Seems like a problem with the library, how can I fix it please?

Thank you.

Source: https://learn.sparkfun.com/tutorials/rgb-panel-hookup-guide

ladyada, 2019-09-10:
what hardware are you using

peverett, 2019-09-11:
M'Lady,

I have the same issue although it may not be that the clock is 'zero'.

Hardware:
- Adafruit Metro 328 - Powered by USB from my laptop
- Adafruit RGB Matrix Shield (prebuilt).
- 32x64 RGB 'hub75' pixel board - Powered from a 5v 4a brick

Software:
- Arduino IDE verson 1.8.9
- Library: RGB matrix panel 1.10 installed through the IDE

Using the example sketches supplied with the library, I have observed the following;

Compiling and flashing the testshapes_32x32.ino sketch and executing it on the above configuration is successful. The image is duplicated on both sides of the panel (32x32 + 32x32). The left hand copy has some ghosting.

I have no scope but using a multimeter can read 0.10 V on Pin 8 (measured from the via next to the pin and via for ground). I assume this value may be because the low speed clock signal is similar in effect to PWM.

Compiling and flashing the testshapes_32x64.ino sketch and executing results in no output on the LED panel. There is also a compiler warning:
```
C:\Users\Simon\Documents\Arduino\libraries\RGB_matrix_Panel\examples\testshapes_32x64\testshapes_32x64.ino: In function 'void setup()':

C:\Users\Simon\Documents\Arduino\libraries\RGB_matrix_Panel\examples\testshapes_32x64\testshapes_32x64.ino:68:15: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]

   char *str = "AdafruitIndustries";
               ^
```
Again, reading the voltage at Pin 8 produces a value of 4.19 V, this is as close to 'ON' as most Digital outputs get for Arduino in my limited experience, but this is my first project with the Metro 328.

Kind Regards

Pev

ladyada, 2019-09-11:
you cant have 64 pixel wide display on 328p

rtek1000, 2019-09-11:
@ladyada ESP8266 (ESP12)

ladyada, 2019-09-11:
this code does not support ESP8266

closed 2019-09-11 by ladyada

rtek1000, 2019-09-11:
@ladyada 

Sorry, for this code the errors happened with Atmega328,

I am using ESP32 and ESP8266 to test the panel

But I see "you cant have 64 pixel wide display on 328p"

Thank you.