adafruit/RGB-matrix-Panel #68: ESP32 + WiFi + matrix panel

pilnikov, 2021-07-27:
I get your example [scrolltext_16x32](https://github.com/adafruit/RGB-matrix-Panel/tree/master/examples/scrolltext_16x32) as a basis. 
Standalone it works great. 
But .... 
I tried to add WiFi client, just one line - 
`WiFi.begin (ssid, password);`
Got this:
`09:00:59.135 -> Connecting to MyWiFi `
`09:00:59.135 -> Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed). `
`09:00:59.135 -> `
`09:00:59.135 -> Core  0 register dump: `
`09:00:59.181 -> PC      : 0x400d5184  PS      : 0x00060034  A0      : 0x80081e78  A1      : 0x3ffbeaa0  `
`09:00:59.181 -> A2      : 0x3ffc1b4c  A3      : 0x3ff44004  A4      : 0x3ffc1f4c  A5      : 0x000001ae  `
`09:00:59.181 -> A6      : 0x08840004  A7      : 0x3ffc253c  A8      : 0x80081234  A9      : 0x00000004  `
`09:00:59.181 -> A10     : 0x0000000a  A11     : 0xff000000  A12     : 0x80083060  A13     : 0x3ffaf980  `
`09:00:59.181 -> A14     : 0x00000020  A15     : 0x13002248  SAR     : 0x00000020  EXCCAUSE: 0x00000007  `
`09:00:59.228 -> EXCVADDR: 0x00000000  LBEG    : 0x400d905c  LEND    : 0x400d9077  LCOUNT  : 0x00000004  `
`09:00:59.228 -> `
`09:00:59.228 -> `
`09:00:59.228 -> Backtrace:0x400d5181:0x3ffbeaa00x40081e75:0x3ffbeac0 0x40085505:0x3ffbeae0 0x4008f1f4:0x3ffaf970 0x40091ce1:0x3ffaf9a0 0x40092467:0x3ffaf9c0 0x40082a59:0x3ffafa40 0x400d6c4a:0x3ffafa80 0x400d8b82:0x3ffafab0 0x400d9685:0x3ffafad0 0x400d9a3f:0x3ffafaf0 0x400d884b:0x3ffafb70 0x400d8aad:0x3ffafbe0 0x400d6f09:0x3ffafc00 0x400d76f1:0x3ffafc30 0x4011cbca:0x3ffafc50 0x4011cd83:0x3ffafc90 0x4011a910:0x3ffafcb0 0x401195a8:0x3ffafcd0 0x4014ccbd:0x3ffafcf0 `
`09:00:59.275 -> `
`09:00:59.275 -> `
`09:00:59.275 -> `
`09:00:59.275 -> `
`09:00:59.275 -> ELF file SHA256: 0000000000000000`
`09:00:59.275 -> `
`09:00:59.275 -> Rebooting...`

P.S. I also had to change two lines from here:
[RGBmatrixPanel.cpp](https://github.com/adafruit/RGB-matrix-Panel/blob/master/RGBmatrixPanel.cpp)
`line 347   tim_config.alarm_en = true;`
`line 348   tim_config.auto_reload = true;`
on the
`line 347  tim_config.alarm_en = TIMER_ALARM_EN;`
`line 348  tim_config.auto_reload = TIMER_AUTORELOAD_EN;`

Tnx, Alex

PaintYourDragon, 2021-07-27:
This library is somewhat deprecated now in favor of Adafruit_Protomatter, does the same issue exist when using that code?

pilnikov, 2021-07-27:
if i use protomatter - all working fine in any compositions, but....
I specifically wanted to use one library for atmega2560 and esp32. Evidently I will have to use different for each MCU

closed 2021-07-27 by pilnikov