adafruit/RGB-matrix-Panel #69: Error with panel Hl1.0

soleyj, 2022-06-03:
We have an Arduino mega that works with a rgb pannel 64x32 D1.0 version. 

But now we have bought a new panel that is version HL1.0 and the system stop working.  

We are using the RGBmatrixpanel library. 

I have seen this post (https://forums.adafruit.com/download/file.php?id=46238)  that have the same problem but after all is no solution. They said that may be a problem with the clock speed, do you know where I can try new values and which are these values.

Thanks!

ladyada, 2022-06-03:
where precisely did you purchase it? we only have tested adafruit rgb matricies - if not purchased from us we have no way to test and you'll need to contact the vendor they can provide techsupport!

caternuson, 2022-06-03:
Cross posted in forums:
https://forums.adafruit.com/viewtopic.php?f=47&t=191687