adafruit/RGB-matrix-Panel #70: Runtime Exception with ESP32

ChaseParate, 2022-06-14:
- Arduino board:  **DOIT ESP32 DEVKIT V1**

- Arduino IDE version (found in Arduino -> About Arduino menu):  **1.8.19**

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too):
    - I've wired up my ESP32 to my official Adafruit 32x16 LED matrix.
    - Using this simple sketch, I get an error in the Serial Monitor:
        ```c
        #include <RGBmatrixPanel.h>

        #define A_PIN 23
        #define B_PIN 19
        #define C_PIN 5
        #define CLK_PIN 16
        #define LAT_PIN 4
        #define OE_PIN 15

        #define R1_PIN 25
        #define G1_PIN 26
        #define B1_PIN 27
        #define R2_PIN 14
        #define G2_PIN 12
        #define B2_PIN 13

        const unsigned int pinlist[] = {R1_PIN, G1_PIN, B1_PIN, R2_PIN, G2_PIN, B2_PIN};
        RGBmatrixPanel matrix(A_PIN, B_PIN, C_PIN, CLK_PIN, LAT_PIN, OE_PIN, false, pinlist);

        void setup() {
          Serial.begin(115200);
          matrix.begin();

          Serial.println("Is anyone there?");
        }

        void loop() {
        }
        ```
    - The aforementioned error is:
        ```
        Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
        Core 1 register dump:
        PC      : 0x40081969  PS      : 0x00060030  A0      : 0x800d101e  A1      : 0x3ffb1f10 
        A2      : 0x00000007  A3      : 0x00000002  A4      : 0x80086a78  A5      : 0x3ffba320 
        A6      : 0x00000001  A7      : 0x40084368  A8      : 0x3f40094c  A9      : 0xffffffff 
        A10     : 0xffffffff  A11     : 0x00000064  A12     : 0x08000000  A13     : 0x3ffbca00 
        A14     : 0xffffffff  A15     : 0x00000000  SAR     : 0x0000001a  EXCCAUSE: 0x0000001c 
        EXCVADDR: 0xffffffff  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000 

        ELF file SHA256: 0000000000000000

        Backtrace: 0x40081969:0x3ffb1f10 0x400d101b:0x3ffb1f30 0x400d0c54:0x3ffb1f90 0x400d204a:0x3ffb1fb0 0x40086a75:0x3ffb1fd0
        ```
   - Finally, I tried inspecting the backtrace, and got this as a result:
        ```
        Decoding stack results
        0x40081969: __pinMode at C:\Users\Chase\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32\esp32-hal-gpio.c line 106
        0x400d101b: RGBmatrixPanel::begin() at C:\Users\Chase\Documents\Arduino\libraries\RGB_matrix_Panel\RGBmatrixPanel.cpp line 314
        0x400d204a: String::changeBuffer(unsigned int) at C:\Users\Chase\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32\WString.cpp line 167
        0x40086a75: xQueueTakeMutexRecursive at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c line 651
        ```

Since this error seems to be happening in the `RGBmatrixPanel::begin()` function, I thought I'd post here for help.
[I've also created a post on the forums for this.](https://forums.adafruit.com/viewtopic.php?f=47&t=191979)

Thanks!
Chase

marcmerlin, 2022-06-14:
I'm not sure how stable or new the ESP32 code is, but may I recommend this library instead?
https://github.com/pixelmatix/SmartMatrix
If you want GFX support, you can add
https://github.com/marcmerlin/SmartMatrix_GFX

This will also give you 24bppp support

ChaseParate, 2022-06-15:
This seems to be some kind of null-pointer exception. The `ptr` variable in the updateDisplay function is a null-pointer that is trying to be dereferenced.

closed 2022-06-15 by ChaseParate

ladyada, 2022-06-17:
https://github.com/adafruit/Adafruit_Protomatter supports ESP32 but not this library