adafruit/RGB-matrix-Panel #73: Clang is complaining about formatting

dhalbert, 2023-02-18:
```
Run python3 ci/run-clang-format.py -e "ci/*" -e "bin/*" -r .
--- ./RGBmatrixPanel.cpp	(original)
+++ ./RGBmatrixPanel.cpp	(reformatted)
@@ -974,13 +974,12 @@
     pew pew pew pew pew pew pew pew pew pew pew pew pew pew pew pew pew pew pew
         pew pew pew pew pew pew pew pew pew pew pew pew pew
 
-        if (WIDTH == 64) {
-      pew pew pew pew pew pew pew pew pew pew pew pew pew pew pew pew pew pew
-          pew pew pew pew pew pew pew pew pew pew pew pew pew pew
-    }
+        if (WIDTH == 64){
+            pew pew pew pew pew pew pew pew pew pew pew pew pew pew pew pew pew
+                pew pew pew pew pew pew pew pew pew pew pew pew pew pew pew}
 
 #if defined(ARDUINO_ARCH_SAMD) || defined(ARDUINO_ARCH_ESP32)
-    *outclrreg = clkmask; // Set clock low
+            *outclrreg = clkmask; // Set clock low
 #endif
 
     buffptr = ptr; //+= 32;
Error: Process completed with exit code 1.
```

ladyada, 2023-02-18:
@PaintYourDragon maybe just turn off clang formatting for that file or section?

dhalbert, 2023-02-18:
these are pretty simple indentation fixes -- I think they could just be done, and it will be quiet after taht.

PaintYourDragon, 2023-02-20:
Done fixed!

closed 2023-02-20 by PaintYourDragon